PaulStoffregen/RadioHead #1: RH_ASK doesn't work with Servo.h

Isaac96, 2015-04-12:
I can't use the Servo library with Radiohead. I think it's an issue with the timers conflicting. How can I fix this?
Thanks
-Isaac

PaulStoffregen, 2015-04-12:
You're using a Teensy, right?  If so, please post here for help.  Try to follow the "forum rule"....

https://forum.pjrc.com/forums/3-Technical-Support-amp-Questions

If conversation there turns up a bug, please open a new issue here with more specific info about the problem.

closed 2015-04-12 by PaulStoffregen

FrankOcean11, 2015-09-25:
im having the same issue but im on a nano

here is the code i have 

// Include Libraries
include <SPI.h
include <RH_NRF24.h
include <Servo.h

int i=0;
int j=0;
RH_NRF24 nrf24; // Create NRF24 object

uint8_t command[11]; // Create array for data to be rx'ed

uint8_t buf[RH_NRF24_MAX_MESSAGE_LEN]; // Create an array buffer for the NRF24 module

void setup() {

// initialize the radio module
nrf24.init();
nrf24.setChannel(1);
nrf24.setRF(RH_NRF24::DataRate2Mbps, RH_NRF24::TransmitPower0dBm);
Serial.begin(115200);
}

void loop() {
  // put your main code here, to run repeatedly:
 if (nrf24.available()) // If the NRF24 has received data
{
uint8_t buf[11]; // Create array to store it in
uint8_t len = sizeof(buf);
if (nrf24.recv(buf, &len)) // Store the data
{
   Serial.println("Rx data:");
  for(int i=0;i<len;i++)
  {
   Serial.print(buf[i]); 
  }
 Serial.println();
  
}

}}
******************************ERROR****************************************
Arduino: 1.6.5 (Windows 8.1), Board: "Arduino Nano, ATmega328"

RadioHead\RH_ASK.cpp.o: In function `__vector_11':
C:\Program Files (x86)\Arduino\libraries\RadioHead/RH_ASK.cpp:491: multiple definition of `__vector_11'
Servo\avr\Servo.cpp.o:C:\Program Files (x86)\Arduino\libraries\Servo\src\avr/Servo.cpp:81: first defined here
collect2.exe: error: ld returned 1 exit status
Error compiling.

FrankOcean11, 2015-09-25:
all i was trying to do is send values from one mcu to this one that will have a servo on it therefore control my servo over the air

flyerdp, 2017-11-01:
I have the same issue with a Mega, Uno, Mini Pro etc....  Same exact vector error message when including both servo.h and radiohead or the original RF24 libraries.

Isaac96, 2017-11-03:
This is the Teensy fork, so it won't help. forum.arduino.cc