PaulStoffregen/RadioHead #2: Teensy 3.1 not able to init() nrf24l01+

shirish47, 2015-05-24:
I am sure after a week of trying that wires are correctly connected proper voltage (3.24V ) and capacitor 10uF is connected to nrf24. same circuit with pro mini is working as server and client.
Now I am trying to upgrade my project to teensy 3.1 and it fails in init() at 
 if (spiReadRegister(RH_NRF24_REG_1D_FEATURE) != (RH_NRF24_EN_DPL | RH_NRF24_EN_DYN_ACK)){
            return false;

wiring is proper for sure. 
I have tried all possible things for now. 
my arduino is 1.6.3 and teensyduino 1.23.

bundabrg, 2015-08-17:
I have the same issue. Same setup on my Arduino works fine. The teensy 3.1 using either pin 8 or pin 9 for CE (using the appropriate singleton definition) and it does not init.

shirish47, 2015-08-17:
I have given up ... but I think I will check it out over logic analyzer ... and see if things are going good.. those black modules have issue with teensy ..but people have used sparkfun modules without any problem ...what module you are using??

bundabrg, 2015-08-17:
I've been tracking your conversations around the web about this topic whilst I swear at my simple circuitry.

I bought 4 from [here](http://www.ebay.com.au/itm/271412764320)

However in my case it may be my wiring. I swapped my radios between my Arduino and my Teensy with the error still occuring on the Teensy, but in frustration I pulled EVERYTHING off the arduino (cables and all) and plugged them into the Teensy and the thing passed the init finally. So either every time I rewired the Teensy I mucked it up or the wires themselves that I used were too long to too bad quality (I swear I tested with a multimeter that every wire was working before). I'm going to sleep on it and then rebuild the Arduino with the old wires and see what happens.

shirish47, 2015-08-17:
yaa those are the same modules I am using.. let me test this my self.. :) I have a logic analyzer now.. so let me see ..

shirish47, 2015-08-18:
[image]

I think I have done it with not much checking because I had previously spent much of time testing this circuits.. at that time I had no Logic analyzer.. so I just took that circuit tested it again and here is the Image ..  seems that CE is not pulled down here .. Same I tested with a circuit I have built with pro mini but just removed nrf to see it pulls down CE ..but it too does not .. so I think its an Input from NRF or what I am not sure.. 

but here is another image from another forum... http://www.avrfreaks.net/sites/default/files/mirf.png

bundabrg, 2015-08-18:
Assuming Channel 3 in your analyser (Man I gotta get myself one of them) is 3 on the NRF24 then yeah that line is flat. But that would suggest that it would fail with both the Teensy and Arduino where as your OP mentions the Pro Mini works fine.

BTW, I rebuilt my circuit and now I have the arduino failing and my Teensy working so now I'm wondering if its my wiring being too noisy (I'll try stick my oscilloscope and check it out when I get some kid free time) and will attempt to rebuild both with short ribbon cable direct to the headers.

shirish47, 2015-08-18:
Actually my promini one works but today I tested without module and just LA .. to see that promini pulls down CE ? but it also doesn't pull down CE. something  I didn't get. the other pic in the above post is from some other forum.. it show the CE is being pulled down ..

bundabrg, 2015-08-18:
Hmm, CE would be required to enable it so my gut feeling is if the CE is not pulled down then theres something wacky going on. Can you write a simple program to just pull it up/down to make sure your connection is ok? Unless I'm missing something, no CE means pro-mini should not be working as well.

One option would be to pull it down yourself with a resistor to ground (as a test) to force it down and see if everything starts working (Assuming nothing else shares SPI).

shirish47, 2015-08-18:
Ya for testing purpose I had removed nrf from schematic but I am sure NRF works with promini... I also pulled down CE in case of teensy 3.1 but nothing happened I think I would take some free time to debug this ...right now busy with other things .. or else I would have made a new circuit .. :)
:+1:

bundabrg, 2015-08-23:
I must withdraw my reports of issues with the Teensy. After replacing all my jumpers with better wiring, soldering a new header to my NRF24's to allow it to be plugged directly into my breadboard and adding a 22uF decoupling capacitor I now get init and communication working between a mega 2560 and Teensy 3.1. I also dropped the Teensy to 24Mhz (down from its default of 96Mhz optimized overclocked).