PaulStoffregen/RadioHead #9: RH95 driver not interrupt safe on ESP8266

jcmg92, 2016-10-23:
I am playing around with SX1296 modules connected to ESP8266's and found out that the interrupt handlers called within the various layers of drivers (RH95, various SPI layers, ...) are not ESP8266 interrupt safe (they do not have the ICACHE_RAM_ATTR).

I also had to modify Arduino's ESP8266 core SPI library as I also found this one "not interrupt safe". 

For now, I have made my own mods and seem to now have a stable ESP8266 implementation (scope to RH95 for now).

Am I the only one to have met those issues?

If any interrest, I am happy to share the mods even if they are limited to RH95 support.

aimanhakimzahari, 2020-10-23:
Hi jcmg92, I also faced the same problem. Do you mind if you share your modifications with me? Here is my email aimanzahar1998@gmail.com. Thank you!

CarloRoma63, 2023-12-20:
Same problem. Solved editing radiohead.h at line 1684 replacing ICACHE_RAM_ATTR with IRAM_ATTR

PaulStoffregen, 2023-12-21:
This repository is a copy meant for Teensy.  This is absolutely the wrong place to discuss problems specific to Espressif chips.

I'm not denying a problem exists.  I'm only saying this is the wrong place to discuss, as nothing will come of anything said here.

closed 2023-12-21 by PaulStoffregen