PaulStoffregen/RadioHead #11: Servo.h ahd RH_RF2.h multiple definition of `__vector_11'

tsanchev, 2017-02-02:
this is what is causing the problem
#include <SPI.h>
#include <RH_RF22.h>
#include <Servo.h>

this is what error i've got:

libraries\Servo\avr\Servo.cpp.o (symbol from plugin): In function `ServoCount':

(.text+0x0): multiple definition of `__vector_11'

libraries\RadioHead\RH_ASK.cpp.o (symbol from plugin):(.text+0x0): first defined here

collect2.exe: error: ld returned 1 exit status

exit status 1
Error compiling for board Arduino Pro or Pro Mini.

PaulStoffregen, 2017-02-02:
Yup, they conflict, needing the same hardware resource.  You simply can't use both of these on that board.

Your options are PWMServo, or switching to a more capable board with more hardware resources that don't conflict.

closed 2017-02-02 by PaulStoffregen

flyerdp, 2017-11-01:
Does a "more capable board" include, leanardo, uno, mega ?  They ALL do that for me :/