PaulStoffregen/RadioHead #12: RH_RF69 send hang due to waitPacketSent failed

Pixinger, 2017-02-24:
Hi
I have problems with stable transmissions for the RFM69. Initialization seems to be fine, but when I transmit with `_spRFM->send(_spData, 2)` sometimes the call will never come back. It looks like the internal call `waitPacketSend()` is the reason. Does someone know what the reasons are for not returning from waitPacketSend? The comparison inside is `while (_mode == RHModeTx)`. But what is responsible for not "leaving" the transmission-mode? Can it be that it is something like too much noise on the radio frequency or no receiver found? As far as I understand it is not a confirmed transmission, it is fire and forget?
Has someone an idea?
Kind regards 
Pix

antonmeyer, 2017-07-12:
_mode is set in the interrupt routine. Did you connect the right Pin to the interrupt?

sebasira, 2017-08-17:
Hi! Did you solve it? I'm having a similar issue with RF95 at the waitPacketSend.

I'm using and Arduino Pro Mini Clone with a LoRa Bee. You talk about an interrupt pin. Which one is it? Where should be connected to the Arduino?

Right now I have only wired the SPI and D10 as ChipSelect. Which pin of the LoRa Bee is the interrupt?

Thank you!

antonmeyer, 2017-08-17:
it was a simple typo: 686 -> 868 MHz

antonmeyer, 2017-08-17:
I have no LoRa Bee

sebasira, 2017-08-17:
Thanks! I came back to post I've found the solution to my problem at I see your new answer. 

Indeed, DIO0 is the interrupt that should be attached to pin 2 of Arduino.

This artivle was very helpful: http://cpham.perso.univ-pau.fr/LORA/LoRaDevices.html