PaulStoffregen/RadioHead #13: NRF24L01 Packet Queueing Problem

JamesLenz, 2017-06-02:
I am trying to create a simple TCP-like sending protocol by sending, waiting for an ack, and if no ack is received in a certain amount of time, try again.

I got it mostly working except for a frequent, but random occurring problem. The recipient will randomly miss a packet, but when the next packet is received, it receives the previous missed packet. I have tried clearing and flusing the rx buffer, but to no avail. It appears as if the rx queue is receiving it, but it cannot be detected until another packet is received.

Perhaps a packet is available, but available() is returning an incorrect value?

Here is the output that demonstrates this issue
[image]

dangerverma, 2017-12-07:
Don't ask my why, i am just a technology enthusiast but when i put in 

TX:
nrf24.setChannel(1);
(in setup())

and 
RX:
nrf24.setChannel(5);
(again in setup())

it works brilliantly and consistently each time without any queuing.

Some radio genius should be able to answer it.

Try and see if it works for you!

JamesLenz, 2017-12-07:
I will probably give it another go after finals, I have NO time to mess with this right now
but if this does indeed fix the problem, thank you

iwasz, 2018-01-14:
I had similar problem. I wired two nRFs to a STM32F4 board, where one was sending and the other receiving and I compared what was sent with the data received. And to my surprise data I was receiving sometimes would arrive in wrong order (BTW the same code on much slower STM32F0 board was fine)! My problem proved to be due to wrong CE pin code in my transmit function (I use my own nRF lib). So faulty code looked like that :

```
void Nrf24L01P::transmit (uint8_t *data, size_t len)
{
        setCe (true);
        spi->setNss (false);
        spi->transmit8 (W_TX_PAYLOAD);
        spi->transmit8 (data, len, nullptr, bogoDelay);
        spi->setNss (true);
        setCe (false);
}
```
And the code that fixed my problem :

```
void Nrf24L01P::transmit (uint8_t *data, size_t len)
{
        spi->setNss (false);
        spi->transmit8 (W_TX_PAYLOAD);
        spi->transmit8 (data, len, nullptr, bogoDelay);
        spi->setNss (true);
        setCe (true);
        HAL_Delay (1);
        setCe (false);
}
```

Only after the data to be transmitted is put into the RX FIFO, I make a HI pulse on CE pin. Diagram on page 35 of nRF24L01+ pointed me in the right direction. PS of course 1ms is too long and this delay is to be modified. Full code is here : https://github.com/iwasz/libmicro/tree/master/src/rf

rmolinowski, 2023-05-02:
Greetings, great library, but I have this issue as well.  Since I'm sending a joystick control position several times a second over RF, it's not too big an issue, but the "solution" by dangerverma, above makes no sense, and Radiohead uses spiburstwrite and not the method shown by iwasz.  Any assistance on this would be helpful.