PaulStoffregen/RadioHead #17: Don't work in STM32duino

wizz777, 2018-02-24:
Users/wizz/Documents/Arduino/libraries/RadioHead/RHHardwareSPI.cpp: In member function 'virtual void RHHardwareSPI::begin()':
/Users/wizz/Documents/Arduino/libraries/RadioHead/RHHardwareSPI.cpp:117:29: error: invalid conversion from 'uint8_t {aka unsigned char}' to 'BitOrder' [-fpermissive]
     SPI.setBitOrder(bitOrder);
                             ^
In file included from /Users/wizz/Documents/Arduino/libraries/RadioHead/RadioHead.h:739:0,
                 from /Users/wizz/Documents/Arduino/libraries/RadioHead/RHGenericSPI.h:10,
                 from /Users/wizz/Documents/Arduino/libraries/RadioHead/RHHardwareSPI.h:10,
                 from /Users/wizz/Documents/Arduino/libraries/RadioHead/RHHardwareSPI.cpp:7:
/Users/wizz/Documents/Arduino/hardware/Arduino_STM32/STM32F1/libraries/SPI/src/SPI.h:220:7: error:   initializing argument 1 of 'void SPIClass::setBitOrder(BitOrder)' [-fpermissive]
  void setBitOrder(BitOrder bitOrder); 
       ^
exit status 1
Error compile for board Generic STM32F103Z series.

manasdasju, 2018-06-21:
i am also getting similar error when uploading following program:

#include <A_STM32_Examples.h>

#include <RadioHead.h>
#include <RHGenericSPI.h>
#include <RH_NRF51.h>
#include <RHCRC.h>
#include <RHRouter.h>
#include <RH_RF95.h>
#include <RHSoftwareSPI.h>
#include <RHDatagram.h>
#include <RH_CC110.h>
#include <RH_E32.h>
#include <RHHardwareSPI.h>
#include <RH_RF24.h>
#include <RH_RF22.h>
#include <RHSPIDriver.h>
#include <RH_MRF89.h>
#include <RHEncryptedDriver.h>
#include <RHGenericDriver.h>
#include <RH_NRF24.h>
#include <RHReliableDatagram.h>
#include <RHTcpProtocol.h>
#include <RH_Serial.h>
#include <RH_ASK.h>
#include <RHNRFSPIDriver.h>
#include <RH_TCP.h>
#include <RH_RF69.h>
#include <RHMesh.h>
#include <RH_NRF905.h>

/*
  LoRa Simple Client for Arduino :
  Support Devices: LoRa Shield + Arduino 
  
  Example sketch showing how to create a simple messageing client, 
  with the RH_RF95 class. RH_RF95 class does not provide for addressing or
  reliability, so you should only use RH_RF95 if you do not need the higher
  level messaging abilities.

  It is designed to work with the other example LoRa Simple Server

  modified 16 11 2016
  by Edwin Chen <support@dragino.com>
  Dragino Technology Co., Limited
*/

#include <SPI.h>
#include <RH_RF95.h>

// Singleton instance of the radio driver
RH_RF95 rf95;
float frequency = 868.0;

int demo = 4;
void setup() 
{
  pinMode(demo, INPUT); 
  
  Serial.begin(9600);
  while (!Serial) ; // Wait for serial port to be available
  Serial.println("Start LoRa Client");
  if (!rf95.init())
    Serial.println("init failed");
  // Setup ISM frequency
  rf95.setFrequency(frequency);
  // Setup Power,dBm
  rf95.setTxPower(13);
  // Defaults after init are 434.0MHz, 13dBm, Bw = 125 kHz, Cr = 4/5, Sf = 128chips/symbol, CRC on
}

void loop()
{

  uint8_t data[2];
  data[0]=8;

if(digitalRead(demo))

  data[1]=7;
  else
  data[1]=6;
  delay(1);        

   rf95.send(data, sizeof(data));
  
   rf95.waitPacketSent();
 
 
  delay(5000);
}

weingaunity, 2019-11-16:
Hi, i noticed this issue too and i want to use it that way.
Any workarounds by correct #defines up to know?

Thx, Klaus

PaulStoffregen, 2019-11-17:
This is not the official RadioHead.  This is the copy ported for Teensy.

This is not place to report issues with STM or any non-Teensy boards.

closed 2019-11-17 by PaulStoffregen