PaulStoffregen/RadioHead #32: "ISR not in IRAM" crash when RH_RF95.ccp is used with ESP8266 using latest "github.com/esp8266/Arduino" code

Rob58329, 2019-07-06:
When "#include <RH_RF95.h>" is used in the Arduino-IDE with the ESP8266 and any recent "https://github.com/esp8266/Arduino" code (eg.May19), the ESP8266 crashes at run-time at "RH_RF95.init()" with the message "ISR not in IRAM".

I succeeded in fixing this by editing the "RH_RF95.cpp" file, and changing following 3 lines:
from "void RH_RF95::isr0()"  to  "void ICACHE_RAM_ATTR RH_RF95::isr0()"
from "void RH_RF95::isr1()"  to  "void ICACHE_RAM_ATTR RH_RF95::isr1()"
from "void RH_RF95::isr2()"  to  "void ICACHE_RAM_ATTR RH_RF95::isr2()"

I understand that this is an issue for both the ESP8266 and the ESP32 (although I have not tested the ESP32).

I hope this info will be of use!

PaulStoffregen, 2019-07-07:
This is the copy of RadioHead for Teensy.

Do not use it with ESP chips.

Rob58329, 2019-07-08:
@PaulStoffregen: >This is the copy of RadioHead for Teensy

My mistake - I though I was using your library, but I'm actually using "www.airspayce.com/mikem/arduino/RadioHead".

Many thanks for clarifying!

closed 2019-07-08 by Rob58329