PaulStoffregen/RadioHead #38: Error compiling ask_transmitter example on Arduino Nano Every

Freshrik, 2019-12-18:
Hi!

I´m trying to get started with the library on a Nano Every but the interrupt attachments seem "unpopular" with the compiler. First I got this message:
```
Arduino: 1.8.10 (Windows 10), Board: "Arduino Nano Every, None (ATMEGA4809)"

C:\Users\frol\Documents\Arduino\libraries\RadioHead\RH_Serial.cpp:11:11: fatal error: HardwareSerial.h: No such file or directory

Multiple libraries were found for "RH_ASK.h"
  #include <HardwareSerial.h>

 Used: C:\Users\frol\Documents\Arduino\libraries\RadioHead
           ^~~~~~~~~~~~~~~~~~

Multiple libraries were found for "SPI.h"
compilation terminated.

 Used: C:\Users\frol\AppData\Local\Arduino15\packages\arduino\hardware\megaavr\1.8.5\libraries\SPI
exit status 1
Error compiling for board Arduino Nano Every.
```
I then moved the missing files in STM32ArduinoCompat up one level and tried again
Then I got this:
```
Arduino: 1.8.10 (Windows 10), Board: "Arduino Nano Every, None (ATMEGA4809)"

C:\Users\frol\Documents\Arduino\libraries\RadioHead\RHHardwareSPI.cpp: In member function 'virtual void RHHardwareSPI::attachInterrupt()':

C:\Users\frol\Documents\Arduino\libraries\RadioHead\RHHardwareSPI.cpp:70:9: error: 'static void SPIClass::attachInterrupt()' is private within this context

     SPI.attachInterrupt();

         ^~~~~~~~~~~~~~~

In file included from C:\Users\frol\Documents\Arduino\libraries\RadioHead/RadioHead.h:1269:0,

                 from C:\Users\frol\Documents\Arduino\libraries\RadioHead/RHGenericSPI.h:10,

                 from C:\Users\frol\Documents\Arduino\libraries\RadioHead/RHHardwareSPI.h:10,

                 from C:\Users\frol\Documents\Arduino\libraries\RadioHead\RHHardwareSPI.cpp:7:

C:\Users\frol\AppData\Local\Arduino15\packages\arduino\hardware\megaavr\1.8.5\libraries\SPI\src/SPI.h:177:22: note: declared private here

   inline static void attachInterrupt() { SPI0.INTCTRL |= (SPI_IE_bm); }

                      ^~~~~~~~~~~~~~~

C:\Users\frol\Documents\Arduino\libraries\RadioHead\RHHardwareSPI.cpp:70:25: error: 'static void SPIClass::attachInterrupt()' is private within this context

     SPI.attachInterrupt();

                         ^

In file included from C:\Users\frol\Documents\Arduino\libraries\RadioHead/RadioHead.h:1269:0,

                 from C:\Users\frol\Documents\Arduino\libraries\RadioHead/RHGenericSPI.h:10,

                 from C:\Users\frol\Documents\Arduino\libraries\RadioHead/RHHardwareSPI.h:10,

                 from C:\Users\frol\Documents\Arduino\libraries\RadioHead\RHHardwareSPI.cpp:7:

C:\Users\frol\AppData\Local\Arduino15\packages\arduino\hardware\megaavr\1.8.5\libraries\SPI\src/SPI.h:177:22: note: declared private here

   inline static void attachInterrupt() { SPI0.INTCTRL |= (SPI_IE_bm); }

                      ^~~~~~~~~~~~~~~

C:\Users\frol\Documents\Arduino\libraries\RadioHead\RHHardwareSPI.cpp: In member function 'virtual void RHHardwareSPI::detachInterrupt()':

C:\Users\frol\Documents\Arduino\libraries\RadioHead\RHHardwareSPI.cpp:77:9: error: 'static void SPIClass::detachInterrupt()' is private within this context

     SPI.detachInterrupt();

         ^~~~~~~~~~~~~~~

In file included from C:\Users\frol\Documents\Arduino\libraries\RadioHead/RadioHead.h:1269:0,

                 from C:\Users\frol\Documents\Arduino\libraries\RadioHead/RHGenericSPI.h:10,

                 from C:\Users\frol\Documents\Arduino\libraries\RadioHead/RHHardwareSPI.h:10,

                 from C:\Users\frol\Documents\Arduino\libraries\RadioHead\RHHardwareSPI.cpp:7:

C:\Users\frol\AppData\Local\Arduino15\packages\arduino\hardware\megaavr\1.8.5\libraries\SPI\src/SPI.h:178:22: note: declared private here

   inline static void detachInterrupt() { SPI0.INTCTRL &= ~(SPI_IE_bm); }

                      ^~~~~~~~~~~~~~~

C:\Users\frol\Documents\Arduino\libraries\RadioHead\RHHardwareSPI.cpp:77:25: error: 'static void SPIClass::detachInterrupt()' is private within this context

     SPI.detachInterrupt();

                         ^

In file included from C:\Users\frol\Documents\Arduino\libraries\RadioHead/RadioHead.h:1269:0,

                 from C:\Users\frol\Documents\Arduino\libraries\RadioHead/RHGenericSPI.h:10,

                 from C:\Users\frol\Documents\Arduino\libraries\RadioHead/RHHardwareSPI.h:10,

                 from C:\Users\frol\Documents\Arduino\libraries\RadioHead\RHHardwareSPI.cpp:7:

C:\Users\frol\AppData\Local\Arduino15\packages\arduino\hardware\megaavr\1.8.5\libraries\SPI\src/SPI.h:178:22: note: declared private here

   inline static void detachInterrupt() { SPI0.INTCTRL &= ~(SPI_IE_bm); }

                      ^~~~~~~~~~~~~~~

Multiple libraries were found for "RH_ASK.h"
 Used: C:\Users\frol\Documents\Arduino\libraries\RadioHead
Multiple libraries were found for "SPI.h"
 Used: C:\Users\frol\AppData\Local\Arduino15\packages\arduino\hardware\megaavr\1.8.5\libraries\SPI
exit status 1
Error compiling for board Arduino Nano Every.

```
Is my problem that the library isn´t supporting the Every board or am I missing something?

Thank you for any help!

PaulStoffregen, 2019-12-18:
This repository is the version of a library meant for use on Teensy.

For your Nano Every (or any non-Teensy board), get the original library from Mike's website.

http://www.airspayce.com/mikem/arduino/RadioHead/

Or look for a specially ported copy like this one, but specifically targeting your specific board.

closed 2019-12-18 by PaulStoffregen

Freshrik, 2019-12-18:
Sorry, could you specify which repository I misuse? I downloaded the library where your link leads.

Freshrik, 2019-12-18:
Sorry again! Now I see my error here!

ConceptOfVoltage, 2020-02-14:
Hello,
can you please say if this has fixed your problem and if so what you did? As i am having the same problem.
Thanks.