PaulStoffregen/RadioHead #45: RH_ASK.h compiles with NANO, wont compile with NANO EVERY

walshrd, 2021-12-19:
Here is the output from the compiler, IDE 1.8.16.  A NANO EVERY is attached to the USB.  All libraries have been upgraded to current.  If #include <RH_ASK.h> is commented out, (//), the test code runs OK.  Removing the // and an error is displayed by the compiler.  It appears that setBitOrder might have a problem (?).  If an "ordinary" NANO is substituted for the EVERY, the test code runs without any problems.  The IDE is running in a Windows 7 partition within a MacBook running OS Catalina:

\\Mac\Home\Documents\Arduino\libraries\RadioHead\RHHardwareSPI.cpp: In member function 'virtual void RHHardwareSPI::attachInterrupt()':
\\Mac\Home\Documents\Arduino\libraries\RadioHead\RHHardwareSPI.cpp:49:25: error: 'virtual void SPIClassMegaAVR::attachInterrupt()' is private within this context
     SPI.attachInterrupt();
                         ^
In file included from \\Mac\Home\Documents\Arduino\libraries\RadioHead/RadioHead.h:540:0,
                 from \\Mac\Home\Documents\Arduino\libraries\RadioHead/RHGenericSPI.h:10,
                 from \\Mac\Home\Documents\Arduino\libraries\RadioHead/RHHardwareSPI.h:10,
                 from \\Mac\Home\Documents\Arduino\libraries\RadioHead\RHHardwareSPI.cpp:7:
C:\Users\David Richardson\AppData\Local\Arduino15\packages\arduino\hardware\megaavr\1.8.7\libraries\SPI\src/SPI.h:186:15: note: declared private here
   inline void attachInterrupt() { SPI0.INTCTRL |= (SPI_IE_bm); }
               ^~~~~~~~~~~~~~~
\\Mac\Home\Documents\Arduino\libraries\RadioHead\RHHardwareSPI.cpp: In member function 'virtual void RHHardwareSPI::detachInterrupt()':
\\Mac\Home\Documents\Arduino\libraries\RadioHead\RHHardwareSPI.cpp:56:25: error: 'virtual void SPIClassMegaAVR::detachInterrupt()' is private within this context
     SPI.detachInterrupt();
                         ^
In file included from \\Mac\Home\Documents\Arduino\libraries\RadioHead/RadioHead.h:540:0,
                 from \\Mac\Home\Documents\Arduino\libraries\RadioHead/RHGenericSPI.h:10,
                 from \\Mac\Home\Documents\Arduino\libraries\RadioHead/RHHardwareSPI.h:10,
                 from \\Mac\Home\Documents\Arduino\libraries\RadioHead\RHHardwareSPI.cpp:7:
C:\Users\David Richardson\AppData\Local\Arduino15\packages\arduino\hardware\megaavr\1.8.7\libraries\SPI\src/SPI.h:187:15: note: declared private here
   inline void detachInterrupt() { SPI0.INTCTRL &= ~(SPI_IE_bm); }
               ^~~~~~~~~~~~~~~
\\Mac\Home\Documents\Arduino\libraries\RadioHead\RHHardwareSPI.cpp: In member function 'virtual void RHHardwareSPI::begin()':
\\Mac\Home\Documents\Arduino\libraries\RadioHead\RHHardwareSPI.cpp:92:29: warning: invalid conversion from 'uint8_t {aka unsigned char}' to 'BitOrder' [-fpermissive]
     SPI.setBitOrder(bitOrder);
                             ^
In file included from \\Mac\Home\Documents\Arduino\libraries\RadioHead/RadioHead.h:540:0,
                 from \\Mac\Home\Documents\Arduino\libraries\RadioHead/RHGenericSPI.h:10,
                 from \\Mac\Home\Documents\Arduino\libraries\RadioHead/RHHardwareSPI.h:10,
                 from \\Mac\Home\Documents\Arduino\libraries\RadioHead\RHHardwareSPI.cpp:7:
C:\Users\David Richardson\AppData\Local\Arduino15\packages\arduino\hardware\megaavr\1.8.7\libraries\SPI\src/SPI.h:171:8: note:   initializing argument 1 of 'void SPIClassMegaAVR::setBitOrder(BitOrder)'
   void setBitOrder(BitOrder order);
        ^~~~~~~~~~~
exit status 1
Error compiling for board Arduino Nano Every.

PaulStoffregen, 2021-12-19:
You're asking in the wrong place.  This copy of RadioHead is only for Teensy boards.

walshrd, 2021-12-19:
So do I repost this elsewhere???

PaulStoffregen, 2021-12-19:
> So I repost this elsewhere???

Yes.  I can not tell you where you should ask for Nano Every help, only that this is not the place to do so.

closed 2021-12-19 by PaulStoffregen